lardemua/SmObEx #1: Robot is not the right one

joao-pm-santos96, 2019-03-10:
I'm using the fanuc package but it only includes the m6ib and not my variant (m6ib/6s) which has a shorter arm. What modifications do I have to do in order to correct this? A simple change in the urdf, or a completely new CAD?

In the screenshot the tf should be 0.430 instead of 0.615.

 Thanks! 

@miguelriemoliveira 
-------
Robot measures: http://motioncontrolsrobotics.com/downloads/techdocs/M6_RJ3iB.pdf

[image]

joao-pm-santos96, 2019-03-10:
Found out a tutorial on how to create the URDF (wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot). Now just need the CAD

miguelriemoliveira, 2019-03-10:
Hi João.  I can help on tuesday morning. You mus changw the urdf and the cad as well

miguelriemoliveira, 2019-03-10:
and I like your readme, a lot of stuff there already.

joao-pm-santos96, 2019-03-11:
Yeah sure! Justo to update, I'm able to run ROS on the lrmate 200iD (smaller one), now working on the bigger one

closed 2019-03-11 by joao-pm-santos96