lardemua/SmObEx #4: Calibration not working, always tending to the same wrong solution

joao-pm-santos96, 2019-03-15:
Hi @miguelriemoliveira, updated the calibration process like we talked two days ago, but when trying to calibrate, the process always tends to the same wrong solution... already compared to @rarrais issue https://github.com/jhu-lcsr/aruco_hand_eye/pull/5 bue didn't changed much... any suggestion? If I hand publish a "calibration" tf of xyz=0 0 0  and rpy=0 0 0 it lays exactly where it should be.
Pictures show the real system and the badly calibrated one. 
Thanks!

[image]

[image]

miguelriemoliveira, 2019-03-15:
Hello @joao-pm-santos96 ,

I will try to find some time to go to LAR today. Perhaps at 14h, but I am not sure I will have time. 

If I cannot we talk on Monday ...

joao-pm-santos96, 2019-03-15:
Thank you! Really appreciate it

joao-pm-santos96, 2019-03-15:
Found my error, when calling calibration_mode the camera_rgb_optical_frame doenst't match its place... now just have to find where I messed up.

[image]

miguelriemoliveira, 2019-03-15:
Why is it solved? Is it working?

joao-pm-santos96, 2019-03-15:
Yeah seems to be working fine now. Only store_calibration.py not doing what was expected, working on it now

joao-pm-santos96, 2019-03-15:
Now everything works fine! Thanks for the help, will show it to you next monday!

miguelriemoliveira, 2019-03-15:
Hi @joao-pm-santos96 , You can close the issue. I just reopened to try to say that, 
whenever you close an issue, you should explain why you close it. For example, this is now working in commit ca39b3eef52d1d1482b26254ab1d89e0430885f5

joao-pm-santos96, 2019-03-16:
Hello! Ok understood that, will do from now on. Thanks for the help and suggestions!

closed 2019-03-16 by joao-pm-santos96