lardemua/SmObEx #7: Enhancements

joao-pm-santos96, 2019-03-21:
Hello @miguelriemoliveira! What do you think about this? http://wiki.ros.org/pcl_ros/Tutorials/PassThrough%20filtering

It seams to have an higher rate than point_cloud_spatial_filter, and I could still use point_cloud_spatial_filter to define the region and then the nodelet (not sure waht that is...) in the "operation mode". Is it worth the try? Thanks!

p.s. Will create a issue related to the problems with the fanuc.

miguelriemoliveira, 2019-03-21:
I think i have used passthroufh filter i spatial filter ... can you check
please?

joao-pm-santos96, 2019-03-21:
I've checked and I don't think you've used it, at least I didn't found it in neither the .h on launch files

joao-pm-santos96, 2019-04-08:
Hello @miguelriemoliveira and @rarrais . So, following our meting today I've encountered a question: my node will generate all the poses and evaluate them, correct? I need to subscribe /octomap_full, and that subscription will trigger a callback. But most of the times, there is a new message in /octomap_full without having any movement of the camera. My question is, if the callback is called every time there is a new message to be read, and the poses are generated inside the callback, the poses will change every time before the camera even moves. So I can't do that... but I need the pose evaluater to access the current state of the octomap... So is there an away to handle the situation inside a single node? Or should I divide the process in two nodes, one to generate and other to evaluate?

joao-pm-santos96, 2019-04-08:
Personally I think that two nodes is the best away, but maybe I'm not seeing the full picture...

joao-pm-santos96, 2019-04-26:
Hello @miguelriemoliveira and @rarrais 

Autonomous mode is now roughly working but I'm having an issue: sometimes while moving, when the robot it's almost stretched, it aborts the movement and gives some errors. It's close to the joint limits but not quite there, and the urdf/xacro have the right joint limits. It's not a singularity either, and even if it was, since the movement is in joint space, it wouldn't be a problem. As much I'm aware of, the IK solver takes in account the joint limits so I don't understand why it happens... any suggestion?

[image]
[image]

miguelriemoliveira, 2019-04-26:
Hi @João Santos <santos.martins.joao@live.ua.pt> ,

I will try to meet you at 13h or 13h30 today. Are you at LAR?

Please open a specific issue for this problem. It is better.

See you soon,
Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-04-26:
Yes, I will be here at that time

miguelriemoliveira, 2019-04-26:
Ok, see you soon.
Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

closed 2020-03-02 by joao-pm-santos96