lardemua/SmObEx #8: Manipulator doesn't receive commands

joao-pm-santos96, 2019-03-21:
ROS Industrial doesn't move with commands sent from pc.

Problems:
- Controller's version is too old (7.20 compared to the "required" 7.70)
- Compilation isn't easily done
- Tried adapt the driver code but wasn't able to adapt everything, and couldn't test that the adapted parts where working

Possible solutions:
- Updating the controller's software

joao-pm-santos96, 2019-03-21:
Tried as it seams roboguide doesn't run on an Windows VM... I'll install it on a separate pc tomorrow and hope it works. Still haven't found any explanation on how to connect roboguide by ethernet.

miguelriemoliveira, 2019-03-22:
https://www.robot-forum.com/robotforum/fanuc-robot-support-forum/socket-messaging-on-virtual-robot-in-roboguide/

joao-pm-santos96, 2019-03-22:
Thanks!! Still waiting for roboguide to finish installation in the windows pc.

joao-pm-santos96, 2019-03-22:
Finished the installation. All files compiled successfully in version 7.70 without any modification.  Now will try to connect to the linux pc with ros

joao-pm-santos96, 2019-03-22:
@miguelriemoliveira so I finally did it and have a couple of important updates.

1) using the package created by me I wasn't able to move the robot in roboguide.
2) using the M6iB package (not altered) I successfully moved the robot (the M6iB/6S variant) in roboguide. Then tried the same thing in the real one, and didn't move.

The junction of the above two points indicates to me that although there is some problem with my package (working on it next) that isn't the issue and updating the controller to v7.70 will solve the problem. 

Do you need something of my part (pictures, video...)?

miguelriemoliveira, 2019-03-22:
Hi. Good work.

If I understand correctly you were able to use ros industrial with
roboguide compiled for version 7.70, so it seems that if motofil updates
the controller version to 7.70 it should be enough.

Thats it no?

I will talk to Rui then...

joao-pm-santos96, 2019-03-22:
Yup, that's it! Thank you

joao-pm-santos96, 2019-04-03:
Hi @rarrais and @miguelriemoliveira . I have good news! Eng. Gonçalo Carpinteiro came here, updated the controller, and it finally moves thru moveit!

closed 2019-04-03 by joao-pm-santos96

miguelriemoliveira, 2019-04-03:
Great news! We are moving forward!