lardemua/SmObEx #9: Good quality video

miguelriemoliveira, 2019-03-22:
Hi @joao-pm-santos96 ,

If you have time you can try to create a video with better quality about the reconstruction.

Take a look at this [video](https://www.youtube.com/watch?v=7crjJD73PM4&feature=youtu.be) done by @rarrais ...

joao-pm-santos96, 2019-03-22:
Hell @miguelriemoliveira ,

right now I'm working on creating the IK fast plugin to the M6iB/6S, I'll need it sonner or latter... when I'm done with it that will be my priority

miguelriemoliveira, 2019-03-22:
... and you can make sure it works using roboguide right?

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-03-22:
Yup, that's what I'm aiming for

joao-pm-santos96, 2019-03-22:
Solved! @TiagoaTavares found out the problem and the solution

miguelriemoliveira, 2019-03-22:
That's great news ... I just don't understand how you let @TiagoaTavares (the enemy) help you :-)

joao-pm-santos96, 2019-03-22:
Yeah I know that feeling... He had (still has actually) some debts so that helped to relive the pain a little bit

joao-pm-santos96, 2019-03-25:
@miguelriemoliveira @rarrais here goes the latest video showing the reconstruction of a defined volume: 

https://youtu.be/-pPXkNzlAXI

rarrais, 2019-03-26:
@joao-pm-santos96 very good video, keep up the good work!

joao-pm-santos96, 2019-04-30:
Also check comment by @miguelriemoliveira in https://github.com/joao-pm-santos96/SmObEx/issues/6#issuecomment-484166850_

closed 2020-03-02 by joao-pm-santos96