lardemua/SmObEx #10: Object to explore

joao-pm-santos96, 2019-03-27:
Hello @miguelriemoliveira  and @rarrais . Regarding the object to explorer a friend suggested 3D printing a building or something like that, so I asked Eng. Rui Heitor for some information about the possibility of doing it here at DEM, and the price and size limitations. After a little google search I've found this CAD that think would be very good (https://www.3dcadbrowser.com/3d-model/marres-castle).

Is the ideia viable? Thanks!

joao-pm-santos96, 2019-03-27:
And this one too https://grabcad.com/library/hexagon-living-dream-modular-home-with-interior-design-by-mab-mehmet-1

rarrais, 2019-03-27:
@joao-pm-santos96 it could work as a testing component, but I would strongly prefer a more industrial object, aligned with a more realistic scenario. I could send you some 3D models (for 3D printing) of some components being used in the FASTEN project.

miguelriemoliveira, 2019-03-27:
Hello,

I like the idea of the castle, especcially because we don't have good industrial applications. 
@rarrais , please send us the CAD or pictures of the parts you mention ... are they complex enough?

One alternative to 3D printing is to purchase one of these models

[1](https://www.amazon.es/AEDES-Aedes1013-Guimaraes-Castle-modelo/dp/B00XUDNIJE/ref=pd_sbs_0_12?_encoding=UTF8&pd_rd_i=B00XUDNIJE&pd_rd_r=9ef9a548-5099-11e9-bbe9-95133ce665cf&pd_rd_w=EFhRY&pd_rd_wg=622qB&pf_rd_p=015a6fb5-f315-4cbc-9908-0dbd43620b7b&pf_rd_r=ETVTBH923SNJ3B59BSN3&psc=1&refRID=ETVTBH923SNJ3B59BSN3)

rarrais, 2019-03-27:
Hey,

I will send you the CAD models, but they are not too complex. Also, they are confidential, so any result (images or videos) cannot be made public immediately and should be validated by the project consortium. 

I suggest an alternative to the castle idea: why not use a complex part with a more industrial look? Examples: a car engine, an alternator, a starter motor, or a piece of mechanical equipment of some kind. Maybe you have something around the lab which can be used and looks a little less out of place.

miguelriemoliveira, 2019-03-27:
Hi @rarrais,

a car engine or a complex machine of some sort sounds like a good idea as well, @joao-pm-santos96 can you find something of the sort here at DEM?

joao-pm-santos96, 2019-03-27:
Tomorrow morning I'll ask around and find out if there's something and will let you know

joao-pm-santos96, 2019-04-30:
This the current scenario: 

[image]

miguelriemoliveira, 2019-04-30:
Looks great ... i cant wait to see the robot exploring the scene...

closed 2020-03-02 by joao-pm-santos96