lardemua/SmObEx #11: Remapping of pcl filter params

joao-pm-santos96, 2019-04-02:
Good morning Professor @miguelriemoliveira . In [Auto mode launch file](./smobex_bringup/launch/auto_mode.launch) I hasn't able to do a remapping  of the point_cloud_spatial_filter params to the ones requested by pcl_filters. Can you give me some help? Thanks

miguelriemoliveira, 2019-04-02:
Hi @joao-pm-santos96 ,

I can help but it should take some days until I have some time.

If I get some free time I will go to LAR, but this week I am very very busy so I make no promisses...

miguelriemoliveira, 2019-04-02:
This Is not hampering the development of the other fucntionalities, is it?

joao-pm-santos96, 2019-04-02:
Hi! No, not at all, this is not urgent, is just to combat the need to press a button to save the BoundingBox params and run a seprate node, but it's no an urgent problem at all

joao-pm-santos96, 2019-04-16:
Done using ros::topic::remap

closed 2019-04-16 by joao-pm-santos96