lardemua/SmObEx #12: Evaluation proceadure

joao-pm-santos96, 2019-04-10:
Hello @miguelriemoliveira and @rarrais .
I just ended the implementation of the method I thought was better to evaluate each pose, which I will describe:
1. generate the 640x480 depth "pixels" (for know they are in sphere coordinates with radius 0.8)
2. for each point calculate (using octomap's castRay) the end point of the ray, either the coordinates of the first occupied voxel or some coordinate in the map's border.
3. for that ray (from start to end of it) calculate the number of unknown voxels intersected

and that's it. It works great for a small number of rays, yet for all the 640*480=308480 pixels it's running for some minutes now (for only one pose..) and it didn't finish yet. 

For today my work is done, tomorrow I'll think in some solution.. If you have any idea or suggestion, please share it with me.

Thanks!

joao-pm-santos96, 2019-04-10:
Actually, step 3 as like a step 3.5 that is running all voxels intersepted by each ray and increment a variable if a voxel is unknown

miguelriemoliveira, 2019-04-10:
Hi @joao-pm-santos96 

Good work. So from what I gather it seems to be working only very slow, correct? Can you mention the code lines we are talking about (you can put a link to them)?

Parallelization is one possibility. Another is to travel only a subset of the pixels. For example, cast a ray only every 10 pixels ...

I did not get step 1 .... Perhaps I ask you to explain it when we meet.

Regards,
Miguel

miguelriemoliveira, 2019-04-10:
Changed the name of the issue to a more explicit title, I think ... change back if you think its not ok.

joao-pm-santos96, 2019-04-21:
Done in commit bfbe04f61b5adc608a0067ed5f3b1fbba0ef7f0b

closed 2019-04-21 by joao-pm-santos96