lardemua/SmObEx #14: Get the center of the unknown clusters

joao-pm-santos96, 2019-04-11:
Hi! If I have one, two or more clusters of unknown space, how can I tell that they are separate clusters and after that, how do I calculate it's center?

miguelriemoliveira, 2019-04-11:
Hi

I have done this before ... search octomap tools... i think it is there.

joao-pm-santos96, 2019-04-14:
Hello!

Found it in [compare_octrees.cpp](https://github.com/miguelriemoliveira/octomap_tools/blob/master/world_model_consistency_check/src/compare_octrees.cpp). Don't know if it is a direct implementation to the SmObEx case but at least I now know that there is an algorithm for that :smile:

joao-pm-santos96, 2019-04-24:
Hello!

Since I have a topic publishing a PointCloud2 that contains the center points of the unknown voxels and I have the octomap's resolution, I should be able to use [pcl's Euclidean Cluster Extraction](http://www.pointclouds.org/documentation/tutorials/cluster_extraction.php#cluster-extraction) to fins the centroid of the unknown clusters, right?

joao-pm-santos96, 2019-04-26:
Done with pcl

closed 2019-04-26 by joao-pm-santos96