lardemua/SmObEx #15: Evaluation of pose too slow with step 1

joao-pm-santos96, 2019-04-15:


joao-pm-santos96, 2019-04-20:
Hello @miguelriemoliveira . 

Remember when we added "#pragma omp parallel for" to the evaluation loop? Well after all that wasn't working because there where missing some libraries on CMakeLists.txt and the compiler was just ignoring that line. They are now added and now I have a core dumped issue... the core is dumped (is this how it is said?) when I use octomap's search(...) function (in line 326 of [explorer.h](https://github.com/joao-pm-santos96/SmObEx/blob/master/smobex_explorer/include/smobex_explorer/explorer.h)). Have you ever had this problem and have any suggestion on how to solve it? 

I just found where the problem is, so I haven't search very much. If I found a solution I'll let you know,  but if you have any idea why this happens and/or how to solve it before I do, please tell me.

Thanks.

miguelriemoliveira, 2019-04-20:
Hi, just a tip: you can add a hyperlink to a line in a code file, e.g., 

https://github.com/joao-pm-santos96/SmObEx/blob/27b5afb40cbddcf6f4c654d55f63bbfad5a21554/smobex_explorer/include/smobex_explorer/explorer.h#L326

I will try to check it, but am not sure what the problem is ...

miguelriemoliveira, 2019-04-20:
Hi again,

can you replace this for loop 

https://github.com/joao-pm-santos96/SmObEx/blob/27b5afb40cbddcf6f4c654d55f63bbfad5a21554/smobex_explorer/include/smobex_explorer/explorer.h#L324

with a normal for loop that does not use iterators?

Does it solve the problem?

joao-pm-santos96, 2019-04-20:
I don't think I can because the KeyRay class as no other way to access the OcKeys. I'm trying this https://stackoverflow.com/questions/2513988/iteration-through-std-containers-in-openmp

joao-pm-santos96, 2019-04-20:
well maybe the problem is not in search(...) after all, I need to investigate more

joao-pm-santos96, 2019-04-20:
Seams to be working now.  KeyRay ray_keys; needed to be defined inside the loop and not outside.

joao-pm-santos96, 2019-04-20:
All of this was to evaluate the time taken to evaluate a pose, wheres waht I found: the method calls ros::topic::waitForMessage("/octomap_full",...) (also wait for the unknown octomap but it's not important for what I'm gonna explain) and this process takes ~0.6/0.7 seconds. The parallel for loop in which the rayCast is done takes about 0.2 or 0.3 seconds... so the bulk of the time is in waiting for an octomap message. Octomap_server is publishing at 273 hz.... any way to improve the rate?

joao-pm-santos96, 2019-04-20:
All of this was to evaluate the time taken to evaluate a pose, here is what I found: the method calls ros::topic::waitForMessage("/octomap_full",...) (also wait for the unknown octomap but it's not important for what I'm gonna explain) and this process takes ~0.6/0.7 seconds. The parallel for loop in which the rayCast is done takes about 0.2 or 0.3 seconds... so the bulk of the time is in waiting for an octomap message. Octomap_server is publishing at 2/3 hz.... any way to improve the rate?

miguelriemoliveira, 2019-04-20:
Hi @joao-pm-santos96 

no, I don't think so.

However you can, everytime you receive an octomap, put it into a class variable and then use this variable as the last received octomap to be processed.

So when we evaluate a pose we just use the last received octomap.

for this you subscribe to the octomap and remove the waitformessage.

Don't forget that the octomap received callback and the evaluate pose functions will collide since one will use the octomap and the other will write on it.

Thus, you need to implement some kind of semaphores or locks mechanism. I think this should do the trick
http://www.cplusplus.com/reference/mutex/mutex/lock/

joao-pm-santos96, 2019-04-20:
Seams a good idea but have a couple of questions:

1) the subscription to the topic is done outside the class, right?
2) will this actually improve speed? Because if the callback is called while evaluating a pose, the evaluation will stop so the callback can be executed and only then the evaluation will be continued right?

miguelriemoliveira, 2019-04-20:
Hi

> Seams a good idea but have a couple of questions:
>
>    1. the subscription to the topic is done outside the class, right?
It can be there or in the class constructor.

>
>    1. will this actually improve speed? Because if the callback is called
>    while evaluating a pose, the evaluation will stop so the callback can be
>    executed and only then the evaluation will be continued right?
But the copy needs onlyto make a copy, not to wait for a new message.
Hence, it should be much faster...

joao-pm-santos96, 2019-04-20:
1. In the class constructor I don't think it's a good idea because I may need to update the maps after the object is created. What I've done is to "manually" call a class method whenever I need to update the maps (for example after moving the robot and before choosing another pose).
2. This way it improved speed just as you said it would.

Thanks!

miguelriemoliveira, 2019-04-20:
Great! So how much time now for evaluating a pose? with which step?

joao-pm-santos96, 2019-04-21:
Between 0.23s and 0.26s with step 20.
Between 0.9s and 1s with step 10. 
And more than one minute and a half with step one...

(Now, when evaluating, the 8 cpu threads are actually all at 100%, so pragma did work.)

joao-pm-santos96, 2019-04-30:
Solved in #20

closed 2019-04-30 by joao-pm-santos96