lardemua/SmObEx #16: Random pose sampling procedure

joao-pm-santos96, 2019-04-19:
Hello @miguelriemoliveira and @rarrais , 

so I was thinking about the best away to generated random valid poses for the camera with tools from MoveIt! and then a question hit me: which is better, to generate random values for the camera position and rotation directly or random joint values? What do you think?

Thnaks!

miguelriemoliveira, 2019-04-19:
Yes, joint values are an interesting approach but they will not comply with the restriction that you want the camera pointed at the center of the target volume ... perhaps a hybrid, some joints (first three) with joint values, some others (last three) to aim the camera?

joao-pm-santos96, 2019-04-23:
My implementation so far computes a random (but reachable) pose and the z orientation is obtained bye the position center point and an observation center (for now, the center of the exploration volume). It works, not very good but also not very bad either. Not very good because some poses would collide with the robot pedestal, not very bad because at least all poses have a score above zero. Any suggestion for improvements?

miguelriemoliveira, 2019-04-23:
Hi @joao-pm-santos96 

lets discuss (very briefly) after Bernardo's presentation this afternoon, and with more detail tomorrow.

joao-pm-santos96, 2019-04-30:
Ended up generating random reachable poses and them imposing the orientation

closed 2019-04-30 by joao-pm-santos96