lardemua/SmObEx #18: Sorting the unknown voxels by distance to camera

miguelriemoliveira, 2019-04-26:
Assume you build a list of class instances, something like this

```cpp

class MyUnknownVoxel
{
   uint32 key;
   center;
   distance_to_camera;

}
```

now with a list of MyUnknownVoxels, you can sort 

```cpp
std::set<MyUnknownVoxel> my_unknown_voxels;
```

using this reference 

https://stackoverflow.com/questions/12823573/c-sorting-class-array

I would imagine you do something like this would work

joao-pm-santos96, 2019-04-27:
Thanks for the guide, the only change was that it couldn't by a set, but using a vector worked out.

closed 2019-04-27 by joao-pm-santos96

miguelriemoliveira, 2019-04-27:
Ok, if you really need a vector you could convert the vector back to a set ... not sure you need it though ...