lardemua/SmObEx #19: Recursively erasing voxels to visit in evaluation

joao-pm-santos96, 2019-04-27:
Hello professor @miguelriemoliveira ,

like we talked yesterday, I'm trying to remove from a vector the voxels that have been visited. Here's my code:

```c++
// for (std::vector<OcTreeKey>::iterator it_vox = keys_to_visit.begin(); it_vox != keys_to_visit.end(); it_vox++)
// {
// 	end_point = unknown_octree->keyToCoord(*it_vox);
while (keys_to_visit.size() != 0)
{
	end_point = unknown_octree->keyToCoord(keys_to_visit[0]);
	unknown_octree->computeRayKeys(origin, end_point, ray_keys);

	bool first = true;
	for (KeyRay::iterator it_key = ray_keys.begin(); it_key != ray_keys.end(); it_key++)
	{
		if (unknown_octree->search(*it_key) && origin.distance(unknown_octree->keyToCoord(*it_key)) >= min_range)
		{
			if (first)
			{
				first_keys.insert(*it_key);
				first = false;
			}
			else
			{
				posterior_keys.insert(*it_key);
			}
		}

		keys_to_visit.erase(std::find(keys_to_visit.begin(), keys_to_visit.end(), *it_key));
	}
	ROS_INFO_STREAM("size " << keys_to_visit.size());
}
ROS_INFO_STREAM("final size " << keys_to_visit.size());
```

and a some point this strange behavior occurs:

```
[ INFO] [1556386765.387367503]: size 1103
[ INFO] [1556386765.388298141]: size 1024
[ INFO] [1556386765.389127382]: size 945
[ INFO] [1556386765.389903400]: size 866
[ INFO] [1556386765.390624482]: size 787
[ INFO] [1556386765.391264054]: size 708
[ INFO] [1556386765.391841766]: size 629
[ INFO] [1556386765.392366459]: size 550
[ INFO] [1556386765.392819317]: size 471
[ INFO] [1556386765.393209733]: size 392
[ INFO] [1556386765.393542126]: size 313
[ INFO] [1556386765.393823790]: size 234
[ INFO] [1556386765.394035866]: size 155
[ INFO] [1556386765.394181466]: size 76
[ INFO] [1556386765.394266308]: size 18446744073709551613
[ INFO] [1556386765.394323962]: size 18446744073709551534
[ INFO] [1556386765.394388536]: size 18446744073709551455
[ INFO] [1556386765.394446115]: size 18446744073709551376
[ INFO] [1556386765.394502303]: size 18446744073709551297
[ INFO] [1556386765.394557470]: size 18446744073709551218
[ INFO] [1556386765.394612997]: size 18446744073709551139
[ INFO] [1556386765.394671177]: size 18446744073709551060
```

Can you see where the problem is? I understand why it becomes an infinity loop, why just can figure out where that jump in size occurs, from 76 to 18446744073709551613... thanks.

joao-pm-santos96, 2019-04-27:
keys_to_visit are already sorted, it's a bad name, should change it.

joao-pm-santos96, 2019-04-27:
Solves with 

```c++
if (keys_sorted.size() == 0)
{
break;
}
keys_sorted.erase(std::find(keys_sorted.begin(), keys_sorted.end(), *it_key));
```

closed 2019-04-27 by joao-pm-santos96

miguelriemoliveira, 2019-04-27:
Hi @joao-pm-santos96 ,

according to my understanding, you cannot use iterators for lists that have dynamic sizes.

Please reformulate the iterator based loop with a simple for loop.

Tell me if that solves it ...

miguelriemoliveira, 2019-04-27:
Did not understand, is it solved?

joao-pm-santos96, 2019-04-27:
Yes, but your suggestion works better, thanks!

miguelriemoliveira, 2019-04-27:
Great!