lardemua/SmObEx #20: New proposed evaluation method

joao-pm-santos96, 2019-04-28:
Hello professor @miguelriemoliveira , I have an update!

so I have good news and bad news. Beginning with the good ones:

The evaluation method that you suggested is implemented and works just as well as you can see in the image bellow (in the image it seams the rays are passing the occupied voxels but it just because of the perspective, it isn't happening). It passes from the 640*480=307200 evals to 1976 (for the pose in the image). 

The bad news is that, for this pose, it took 2 secs. Is an improvement relative to the 1min30sec for the previous method with step 1 and parellisation but with step 20 that time was around 0.2/0.3 secs.

I'll try to improve it some how and give you an update as soon as I have one.

[image]

miguelriemoliveira, 2019-04-28:
Hi ,

I think you are giving results for the first iteration, where the number of
unknown voxels is large.

But as the space becomes known the voxel based method should become faster
and faster, while the fixed rays based method should continue to take the
same ammount of time.

Can you verify this?

I will also look into the code to see if something can be improved.

Good work,
Miguel

joao-pm-santos96, 2019-04-28:
Yes you are right that is for the first iteration, but that's the benchmark iteration because if we want to generate many many poses to evaluate, we cannot afford to have an evaluation time in the seconds.

I've verified and while it's that assumption it's true, the method is still slow. Notice that I haven't done any improvements yet.

Thanks.

joao-pm-santos96, 2019-04-28:
If we can bring down this evaluation time then there is an away to paralelize the pose gen and eval to make it quicker I think

miguelriemoliveira, 2019-04-28:
Can you time profile this approach?  Then we could figure out a way to
improve...

joao-pm-santos96, 2019-04-29:
Yes sure:

Using pcl to get the points inside the frustum, creating and sorting the vector of the unknownVoxel class: 0.36%

The for loop where the ray is casted, keys are removed from the vector and finally keys are added to the evaluation set: 99.63%

joao-pm-santos96, 2019-04-29:
Then inside of this for loop: 

RayCasting, computeRayKeys: 2.9%

Another for loop that distinguishes the keys between first and posterior and deletes (from the vector) the ones that have already been visited: 96.72%

miguelriemoliveira, 2019-04-29:
OK, so we found the culprit.

I will go there in a minute...

Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-04-30:
Done thanks to @miguelriemoliveira

closed 2019-04-30 by joao-pm-santos96