lardemua/SmObEx #21: Robot stops at poses where it is to stretched

joao-pm-santos96, 2019-04-30:
Already tried giving tighter bound limits

miguelriemoliveira, 2019-05-01:
Hi @joao-pm-santos96 ,

did the tightening of the robot's joint limits worked? Did you see the problem again?

joao-pm-santos96, 2019-05-02:
Hello @miguelriemoliveira ,

no it didn't, I happened to me once or twice on tuesday.

joao-pm-santos96, 2019-05-02:
[ERROR] [1556784559.089881692]: Socket sendBytes failed, rc: -1. Error: 'Connection reset by peer' (errno: 104)
[ERROR] [1556784559.100206879]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1556784559.355468608]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1556784559.610696564]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1556784559.866021131]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1556784560.121396369]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1556784560.371564861]: Timeout connecting to robot controller.  Send new motion command to retry.

joao-pm-santos96, 2019-05-02:
Just found out something that must be important!! 

Let's say that the exploration node is running and the robot stops at one of these points. What I normally do to start again is, using MoveIt!, take it to a safe location ans that works just fine, UNLESS I set true tha option "Use Cartesian Path" in Rviz/MoveIt!. Setting it to true gives the exact same error even before it starts moving. So it must be it... 

The question is, nowhere is the code it's used cartesian path... will need to investigate but it's a step forward.

miguelriemoliveira, 2019-05-02:
But is it less often than before? If so you can increase the delta just to
see what happens...
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-02:
I already did that it's not the problem, please see my previous comment 

> Just found out something that must be important!!
>
> Let's say that the exploration node is running and the robot stops at one of these points. What I normally do to start again is, using MoveIt!, [...]

joao-pm-santos96, 2019-05-02:
Seams to be an [Elbow singularity](https://blog.robotiq.com/why-singularities-can-ruin-your-day)

joao-pm-santos96, 2019-05-02:
The working sphere didn't solve it

joao-pm-santos96, 2019-05-05:
Generated IKFAST plugin for the specific robot, using [this tutorial](https://acutronicrobotics.com/docs/products/robots/mara/moveit2/tutorials/ros/ikfast). Add to run part of it on Ubuntu 16.04 in a VM, thought... tomorrow will test to see if improves anything

miguelriemoliveira, 2019-05-05:
Hope it works. ..

joao-pm-santos96, 2019-05-12:
The OpenRAVE (and includes IKFAST) worked out but it still stops in the same kind of poses... tried to use the most recent version of OpenRAVE in the master branch but doesn't even compile because "error: no matching function for call to ‘min(long unsigned int&, unsigned int&)"... working on it but it's taking a lot of time this issue

joao-pm-santos96, 2019-05-16:
I now have a python script that consistently passes in a problematic point

[image]

joao-pm-santos96, 2019-05-16:
The issue seams to be when the elbow wants to change its redundancy, not sure of this just a visual ideia

miguelriemoliveira, 2019-05-16:
a photo of the robot when the problem occurs could be helpful.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-17:
The /joint_states:

```bash
header: 
  seq: 4513
  stamp: 
    secs: 1558098922
    nsecs: 156394777
  frame_id: ''
name: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
position: [1.5252118110656738, 1.3908336162567139, 1.3533990681171417, 1.9825726747512817, -2.0937976837158203, 1.3225916624069214]
velocity: []
effort: []
---
```

The photos:

[image]
[image]
[image]

miguelriemoliveira, 2019-05-17:
Hi @joao-pm-santos96 ,

nice to see you are writing you thesis :-).

Please post the joint limits that were being used, as defined by the xacro file?

joao-pm-santos96, 2019-05-17:
Well @miguelriemoliveira I would feel bad if I had to keep you waiting for those the beautiful photos!!

The xacro values are overwritten by joint_limits.yaml which as the following content: 

```yaml
# joint_limits.yaml allows the dynamics properties specified in the URDF to be overwritten or augmented as needed
# Specific joint properties can be changed with the keys [max_position, min_position, max_velocity, max_acceleration]
# Joint limits can be turned off with [has_velocity_limits, has_acceleration_limits]
joint_limits:
  joint_1:
    has_velocity_limits: true
    max_velocity: 3.4906
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-170.0)
    max_position: deg(170.0)
  joint_2:
    has_velocity_limits: true
    max_velocity: 3.4906
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-90.0)
    max_position: deg(160.0)
  joint_3:
    has_velocity_limits: true
    max_velocity: 4.5378
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-140.0)
    max_position: deg(170.0)
  joint_4:
    has_velocity_limits: true
    max_velocity: 6.9813
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-190.0)
    max_position: deg(190.0)
  joint_5:
    has_velocity_limits: true
    max_velocity: 6.9813
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-140.0)
    max_position: deg(140.0)
  joint_6:
    has_velocity_limits: true
    max_velocity: 12.5663
    has_acceleration_limits: false
    max_acceleration: 0
    min_position: deg(-360.0)
    max_position: deg(360.0)
```

those values are the ones defined in the Teach Pendant

miguelriemoliveira, 2019-05-17:
These are in degrees, the others in radians. Can you check if some joint is outside the limits?

joao-pm-santos96, 2019-05-17:
Yeah sure, so here are the details (WITH THE SIMULATED ROBOT, THAT DOESN'T BREAK): 

For this specific "break point" to happen I send the robot from joint_config _1 to joint_config_2

```python
joint_config_1 = [1.4587480809209663, 0.5394274522857214, 0.26626905547001045, 1.2824433152712542, -2.0201723943348675, 0.7763070202157049]

joint_config_2 = [1.5585755451194119, 1.8164244835560899, 1.8969650689330928, 2.3326661009936167, -2.130728868721306, 1.5956517006099813]
```

and here is a conversions table, as you can see all of the joint_configs are inside the limits.

[image]

When I run the planner, the trajectory that it wants to execute is:

```bash
model_id: "fanuc_xtion"
trajectory: 
  - 
    joint_trajectory: 
      header: 
        seq: 0
        stamp: 
          secs: 0
          nsecs:         0
        frame_id: "base_link"
      joint_names: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
      points: 
        - 
          positions: [1.4587008785290905, 0.5393316407620381, 0.26633590350451525, 1.2824243167785123, -2.0201321210515384, 0.7762564489996674]
          velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
          accelerations: [0.061502324862928996, 0.0, 0.0, 0.0, 0.0, 0.0]
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
          positions: [1.4697952664913025, 0.6812217515956995, 0.44750691478856885, 1.399116698122445, -2.032412902581005, 0.8673034199181727]
          velocities: [0.031539752824981185, 0.4033732225018185, 0.5150431852945241, 0.33173969367952866, -0.03491249947777564, 0.258833472203576]
          accelerations: [0.06154845834445071, 0.7871653313261131, 1.005084415581882, 0.6473756097880099, -0.0681302270101038, 0.5051022837900386]
          effort: []
          time_from_start: 
            secs: 0
            nsecs: 600649204
        - 
          positions: [1.4808896544535146, 0.823111862429361, 0.6286779260726225, 1.5158090794663777, -2.0446936841104715, 0.958350390836678]
          velocities: [0.05127444289411261, 0.65576725908269, 0.8373100619694337, 0.5393120254832356, -0.05675754565032773, 0.420787764673768]
          accelerations: [0.06057098267215758, 0.7746640440129215, 0.9891222681737428, 0.6370943789265362, -0.06704822363842153, 0.4970805524955868]
          effort: []
          time_from_start: 
            secs: 0
            nsecs: 849352978
        - 
          positions: [1.491984042415727, 0.9650019732630224, 0.809848937356676, 1.6325014608103103, -2.056974465639938, 1.0493973617551833]
          velocities: [0.0627397116278181, 0.8024006972591957, 1.024537544747556, 0.659905384561826, -0.06944886859439997, 0.5148783979314138]
          accelerations: [0.0539673897990615, 0.6902083238908536, 0.8812858013943546, 0.5676368315241294, -0.059738466519767316, 0.44288764610713033]
          effort: []
          time_from_start: 
            secs: 1
            nsecs:  40833476
        - 
          positions: [1.503078430377939, 1.106892084096684, 0.9910199486407296, 1.749193842154243, -2.0692552471694046, 1.1404443326736886]
          velocities: [0.07166517612200199, 0.9165516671581457, 1.170290103076957, 0.7537847143612353, -0.07932878985507832, 0.5881259271330279]
          accelerations: [0.053125201246229416, 0.6794372721201469, 0.8675328884504966, 0.558778570202634, -0.05880621738088087, 0.43597615924193683]
          effort: []
          time_from_start: 
            secs: 1
            nsecs: 205098940
        - 
[... 52 lines cut ...]
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "base_link"
    name: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
    position: [1.4587008785290905, 0.5393316407620381, 0.26633590350451525, 1.2824243167785123, -2.0201321210515384, 0.7762564489996674]
    velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    effort: []
  multi_dof_joint_state: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "base_link"
    joint_names: []
    transforms: []
    twist: []
    wrench: []
  attached_collision_objects: []
  is_diff: False
---

```
(MUST CHECK IF NOTHING CHANGES WHEN CONNECTED TO THE REAL ROBOT)

joao-pm-santos96, 2019-05-17:
And the possibilites that I can thing of are:

1. some kind of singularity
2. some problem when joint 3 want to go from elbow up to elbow down
3. some problem related with joint2 and joint3 interaction (less likely since there is only one param to set and it's the correct one that is set, because all others create as mismatch between Rviz and reality)

joao-pm-santos96, 2019-05-17:
a fourth possibility is that, when I generated a .dae to generate the IKFAST c++ code I had to run a command that makes all floating points a 4 decimal place number, will that introduce any problem?

miguelriemoliveira, 2019-05-17:
Don't think so ...
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2019-05-17:
Thanks for the detailed report. All the points in the trajectory are inside
the joint boundaries?

I see many positions there ... we should check all of them are ok ...

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-17:
What if I create a python node that subscribes /joint_states and prints some message if one joint is out of bounds?

joao-pm-santos96, 2019-05-18:
Done the node, seams to always be inside the bounds... Will take on this again next week.

miguelriemoliveira, 2019-05-18:
So it is not a problem of joints out of boundaries ... for sure.

...

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-06-12:
Latest version of IKFast (i.e. openrave) does not solve anything...

joao-pm-santos96, 2019-06-12:
Changing the planner does nothing

joao-pm-santos96, 2019-06-13:
Created a 2.4 bounding cube around the robot, it didn't reached as much but problem showed in other different but similar configuration:

[image]
[image]

miguelriemoliveira, 2019-06-13:
So it did not work?

joao-pm-santos96, 2019-06-13:
No it did not. When can we meet to solve this?

So it did not work?

joao-pm-santos96, 2019-06-14:
Was recording the bag, limited the movement within a cube of 2.4m of side, centered on the base_link. Same error occurred, like this: 

[image]

Tomorrow I'll try to record the bag again

joao-pm-santos96, 2019-06-15:
I think it is fixed. Tested 6 / 7 consecutive times without a single error. The seemingly solution was to add a [jointPathConstraint](http://docs.ros.org/melodic/api/moveit_msgs/html/msg/JointConstraint.html) for joint_3. The errors occurred on joint_3 was out of the bounds -68º to 79º, so I bounded it fo -65º to 75º. For now it seams to work perfectly, forcing new paths to reach the same pose, and actually reaching it.

I'll close the issue, if it occurs again, I'll reopen.

closed 2019-06-15 by joao-pm-santos96

miguelriemoliveira, 2019-06-15:
That is fantastic news!!! SO now you found out the problem! Great.

One thing I did not understand: the console said it was joint 6 with the
error no?

Congratulations!

Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-06-15:
Yes it did, but I had the felling that the problem was on the joint_2 joint_3 interaction. So I looked at that value in a couple of "error poses" and it was practically the same. So I tried to bound the planning to be only inside of those values and, at least for now, it was been working