lardemua/SmObEx #22: Parelization of multiple poses

joao-pm-santos96, 2019-04-30:


joao-pm-santos96, 2019-05-02:
Is it possible to parallelize an simple action server action. the only thing stopping at this point is move group ..?

miguelriemoliveira, 2019-05-02:
Yes, I believe so, you just have to launch a thread to handle the processing of each goal, and accept all of them.

Take a look at the action server example I was doing with Tiago Tavares

https://github.com/miguelriemoliveira/my-action-server

The problem is that this is in python, but he said he could do one in cpp ...

miguelriemoliveira, 2019-05-02:
or perhaps you can use the python code and use cpp only for the octomap stuff ...

joao-pm-santos96, 2019-05-02:
Yeah maybe.. but the problem is in the MoveIt action itself not the one that will be create

miguelriemoliveira, 2019-05-02:
I see, then perhaps it is impossible without rewritting the moveit code
itself.

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-12:
not needed anymore

closed 2019-05-12 by joao-pm-santos96