lardemua/SmObEx #23: Include auto mode on interactive marker menu

joao-pm-santos96, 2019-05-01:
Using actions maybe?

miguelriemoliveira, 2019-05-01:
Hi yes, actions are the way to go. But we should discuss which actions, how thy interact, their granularity, etc ... @rarrais should also give his input here.

Should we have an action for generating a set of poses and evaluating them? Or an action for evaluating a pose? Or just a huge action for exploring the space without any subactions?

@joao-pm-santos96 , can you make a first proposal of the system architecture?

joao-pm-santos96, 2019-05-02:
Good morning @miguelriemoliveira and @rarrais ,

I do think that it should be one huge action because we do have a button to evaluate one pose only chosen by the user, and the have auto mode that generates a bunch of them and evaluates them, so maybe two actions?

Yes @miguelriemoliveira sure, I can try. What would you expect to see in the architecture?

miguelriemoliveira, 2019-05-02:
Hi,

just a hand sketch of the system components and their interactions is fine
...

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-15:
Done on commit bdfd32b387de8e74bb0ebfe47cae93be08408237

[image]

closed 2019-05-15 by joao-pm-santos96