lardemua/SmObEx #24: INESC TEC partnership

joao-pm-santos96, 2019-05-08:
Hi @miguelriemoliveira ! Suggested by @rarrais I'm starting this issue to keep you posted on the developments. 

Yesterday: 

- discussion of the best way to implement the algorithm in a ros action compliant with INESC’s stack.

-installation of the camera drivers and calibration packages 

-beginning of the calibration process (study of the best way to port from MoveIt! to INESC’s driver)

Talk to you soon

joao-pm-santos96, 2019-05-08:
So, the topics done today:

- Extrinsic and Intrinsic calibrations are done
- An accumulated point cloud can be generated
- Volume definition works as expected (but needs discussion with @rarrais about how it should actually be implemented for this project)

What remains is what's related with the automatic mode, this means, the skill. The priority for tomorrow is to install the compatible version of PCL so I can hopefully start coding the skill.

joao-pm-santos96, 2019-05-08:
The topics done today:

- Intrinsic and extrinsic calibration done
- Accumulated point cloud can now be generated
- Exploring volume definition works as expected (but needs discussion with @rarrais for the best approach to this project's goal)

The remaining things are all the ones related with the autonomous part. The priority for tomorrow is the installation of the compatible version of PCL so hopefully I can start coding the skill.

joao-pm-santos96, 2019-05-08:
Today’s topics done: 

- intrinsic and extrinsic calibrations done 
- accumulated point clouds can now be recorded 
- exploration volume definition works as expected (but needs discussion with @rarrais about the best approach regarding the best way to do so)

The things that are still pending are the ones related with the autonomous part of the work. The priority for tomorrow is to have the compatible version of PCL installed

miguelriemoliveira, 2019-05-08:
PLC? What's the problem with the pcl you have?

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-08:
I don't know if it's only me or not, but the comments are all out of order in this issue and I don't know why.

Nevertheless the PCL situation is this: the software developed by INESC TEC that will in the robot also requires some version of PCL and to avoid version mismatch issues we need there version to be installed on the machine

miguelriemoliveira, 2019-05-08:
It also happened to me today ... on a different repository ...

miguelriemoliveira, 2019-05-09:
Hi @joao-pm-santos96 ,

One recommendation: when you develop there at INESC you should perhaps create a new branch of your code that contains all the necessary changes for working with ROS kinetic etc.

joao-pm-santos96, 2019-05-09:
Hello @miguelriemoliveira , I don't think it's needed because everything compiled just fine without any changes. 

The basis of the action are done. This means that everything non-related with the actual movement of the robot is working as they should be. What is missing is publishing the best pose to the tf tree (needed by the move_arm skill), the threshold skill (that will decide if the chosePose->moveArmskills cycle stops or continues) and the integration of all skills.

joao-pm-santos96, 2019-05-13:
Today all the integration was completed. Tomorrow will be a day to check if everything’s ok and film the videos

miguelriemoliveira, 2019-05-13:
Great!

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-05-14:
Everything finished today, I’ll give you more details on out tour this Friday

miguelriemoliveira, 2019-05-14:
What? You are going to make me wait until Friday? Not even a video? :)

rarrais, 2019-05-15:
@joao-pm-santos96 have you added me to the skill repository you created on gitlab?

Also, you forgot to create a repo for the friday_tm_configuration_stack/friday_tm folder, which is inside the friday_ws. 

I created one for you on Gitlab. Please also send me your username on there, so that I can add you as a maintainer.

rarrais, 2019-05-15:
Also, there seem to be two instances of the smobex_ur10 repository installed on the robot:

``` bash
friday@friday:~/friday_ws_exploration/src/smobex_ur10$ cat .git/config
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/joao-pm-santos96/smobex_ur10
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
```

and

``` bash
friday@friday:~/smobex_ur10$ cat .git/config
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	url = https://gitlab.inesctec.pt/joao.p.martins/smobex_ur10.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
```

which one should we keep? The first one, right?

rarrais, 2019-05-15:
Nevertheless, the first one, which is in the proper directory, seems to be pointing to a (private) github repo.

joao-pm-santos96, 2019-05-15:
Hello @rarrais 

I just gave you access. My gitlab username is joao.p.martins

Yes I should have deleted one... they are equal except that  ~/friday_ws_exploration/src/smobex_ur10 is "linked" with the private github repository that I created before having access to gitlab and ~/smobex_ur10 is linked with gitlab. They are equal so, if you don't mind, just delete the first one and substitute it with the second one. Sorry for the mistake...

rarrais, 2019-05-15:
Thanks! I'll do it.

Also, as we discussed yesterday, we will be doing some changes to the robot sw organization today. Your launch files might be broken as a result... I'll do my best to try to fix them. Also, we will probably change the camera support today. As the camera is fixed in the support, I don't expect the need to re-calibrate it.

Nevertheless, if we want to showcase the demonstration running to Miguel on Friday, we might require a few minutes before to check if the launch files are ok. Also, as the robot will move, we also might require to re-setup the objective volume for search.

joao-pm-santos96, 2019-05-15:
Thanks!

Hopefully only friday_exploration's run_exploration.launch will need updates, but if you want we can work on it next friday in a few minutes. Well it depends if the mounting rotation is changed or not but calibration and re-setup are fast and not a problem.

rarrais, 2019-05-15:
@joao-pm-santos96 quick update just to say that I've substituted the github repo by the gitlab repo and compiled everything without any issue. 

Also, I've granted you access to the 'friday_tm' project on gitlab. 

Maybe it's a good idea to include these launch files and configuration files within the smobex_ur10 repo. What do you think? This way, we are speeding things up for the setup on Friday.

joao-pm-santos96, 2019-05-15:
Aren't they already there? I do think so...

closed 2020-03-02 by joao-pm-santos96