lardemua/SmObEx #28: Cant compile OpenRAVE for IKFast plugin generator

joao-pm-santos96, 2019-06-11:
Following [this tutorial](https://acutronicrobotics.com/docs/products/robots/mara/moveit2/tutorials/ros/ikfast) and [this](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html). When building OpenRAVE, gives this error: 

```bash
libopenrave-core/libopenrave0.15-core.so.0.15.0: undefined reference to `boost::this_thread::hiden::sleep_until(timespec const&)'
libopenrave/libopenrave0.15.so.0.15.0: undefined reference to `boost::filesystem::path::m_normalize()'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:109: recipe for target 'src/openrave0.15' failed
make[2]: *** [src/openrave0.15] Error 1
CMakeFiles/Makefile2:614: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

joao-pm-santos96, 2019-06-11:
using all the latest versions, against tutorial

joao-pm-santos96, 2019-06-11:
@miguelriemoliveira have you ever had a similar problem? I've already search openRave issues ang google

joao-pm-santos96, 2019-06-11:
Actually, the error is this:

```bash
In file included from /usr/include/fcl/BV/RSS.h:41:0,
                 from /usr/include/fcl/ccd/motion_base.h:46,
                 from /usr/include/fcl/collision_object.h:45,
                 from /usr/include/fcl/collision.h:43,
                 from /home/joao/git/openrave/plugins/fclrave/plugindefs.h:102,
                 from /home/joao/git/openrave/plugins/fclrave/fclrave.cpp:16:
/usr/local/include/fcl/math/constants.h:131:1: error: expected primary-expression before ‘typedef’
 typedef typename detail::ScalarTrait<S>::type Real;
 ^~~~~~~
/usr/local/include/fcl/math/constants.h:131:1: error: expected ‘}’ before ‘typedef’
/usr/local/include/fcl/math/constants.h:131:1: error: expected ‘;’ before ‘typedef’
/usr/local/include/fcl/math/constants.h:134:18: error: ‘S’ does not name a type
 static constexpr S pi() { return S(3.141592653589793238462643383279502884197169399375105820974944592L); }
                  ^
/usr/local/include/fcl/math/constants.h:137:18: error: ‘S’ does not name a type
 static constexpr S phi() { return S(1.618033988749894848204586834365638117720309179805762862135448623L); }
                  ^
/usr/local/include/fcl/math/constants.h:145:8: error: ‘Real’ does not name a type; did you mean ‘read’?
 static Real gjk_default_tolerance() {
        ^~~~
        read
/usr/local/include/fcl/math/constants.h:151:8: error: ‘Real’ does not name a type; did you mean ‘read’?
 static Real eps() {
        ^~~~
        read
/usr/local/include/fcl/math/constants.h:160:8: error: ‘Real’ does not name a type; did you mean ‘read’?
 static Real eps_78() {
        ^~~~
        read
/usr/local/include/fcl/math/constants.h:166:8: error: ‘Real’ does not name a type; did you mean ‘read’?
 static Real eps_34() {
        ^~~~
        read
/usr/local/include/fcl/math/constants.h:172:8: error: ‘Real’ does not name a type; did you mean ‘read’?
 static Real eps_12() {
        ^~~~
        read
/usr/local/include/fcl/math/constants.h:179:20: error: ‘constants’ does not name a type; did you mean ‘constexpr’?
 using constantsf = constants<float>;
                    ^~~~~~~~~
                    constexpr
/usr/local/include/fcl/math/constants.h:180:20: error: ‘constants’ does not name a type; did you mean ‘constexpr’?
 using constantsd = constants<double>;
                    ^~~~~~~~~
                    constexpr
/usr/local/include/fcl/math/constants.h:182:1: error: expected declaration before ‘}’ token
 } // namespace fcl
 ^
plugins/fclrave/CMakeFiles/fclrave.dir/build.make:62: recipe for target 'plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o' failed
make[2]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o] Error 1
CMakeFiles/Makefile2:2594: recipe for target 'plugins/fclrave/CMakeFiles/fclrave.dir/all' failed
make[1]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

joao-pm-santos96, 2019-06-11:
Installed successfully, needed to install libboost 1.58 on /usr/locl/. Inspired on [openrave-installation](https://github.com/crigroup/openrave-installation) but cmake .. works fine. Will now try to see if makes any difference

joao-pm-santos96, 2019-06-11:
and: https://github.com/rdiankov/openrave/issues/678

miguelriemoliveira, 2019-06-12:
Hi,

sorry, never used openrave ... :-(

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-06-12:
done. results on branch solver_tries

closed 2019-06-12 by joao-pm-santos96