lardemua/SmObEx #29: Record Bag file for results

joao-pm-santos96, 2019-06-13:
List of topics:

```bash
/SmobexExplorerActionSkill/cancel
/SmobexExplorerActionSkill/feedback
/SmobexExplorerActionSkill/goal
/SmobexExplorerActionSkill/result
/SmobexExplorerActionSkill/status
/attached_collision_object
/camera/depth/camera_info
/camera/depth/image
/camera/depth/image/compressed
/camera/depth/image/compressed/parameter_descriptions
/camera/depth/image/compressed/parameter_updates
/camera/depth/image/compressedDepth
/camera/depth/image/compressedDepth/parameter_descriptions
/camera/depth/image/compressedDepth/parameter_updates
/camera/depth/image/theora
/camera/depth/image/theora/parameter_descriptions
/camera/depth/image/theora/parameter_updates
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/depth/image_rect
/camera/depth/image_rect/compressed
/camera/depth/image_rect/compressed/parameter_descriptions
/camera/depth/image_rect/compressed/parameter_updates
/camera/depth/image_rect/compressedDepth
/camera/depth/image_rect/compressedDepth/parameter_descriptions
/camera/depth/image_rect/compressedDepth/parameter_updates
/camera/depth/image_rect/theora
/camera/depth/image_rect/theora/parameter_descriptions
/camera/depth/image_rect/theora/parameter_updates
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/points
/camera/depth_rectify_depth/parameter_descriptions
/camera/depth_rectify_depth/parameter_updates
/camera/depth_registered/camera_info
/camera/depth_registered/image_raw
/camera/depth_registered/image_raw/compressed
/camera/depth_registered/image_raw/compressed/parameter_descriptions
/camera/depth_registered/image_raw/compressed/parameter_updates
/camera/depth_registered/image_raw/compressedDepth
/camera/depth_registered/image_raw/compressedDepth/parameter_descriptions
/camera/depth_registered/image_raw/compressedDepth/parameter_updates
/camera/depth_registered/image_raw/theora
/camera/depth_registered/image_raw/theora/parameter_descriptions
/camera/depth_registered/image_raw/theora/parameter_updates
/camera/depth_registered/points
/camera/depth_registered/sw_registered/camera_info
/camera/depth_registered/sw_registered/image_rect
/camera/depth_registered/sw_registered/image_rect/compressed
/camera/depth_registered/sw_registered/image_rect/compressed/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/compressed/parameter_updates
/camera/depth_registered/sw_registered/image_rect/compressedDepth
/camera/depth_registered/sw_registered/image_rect/compressedDepth/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/compressedDepth/parameter_updates
/camera/depth_registered/sw_registered/image_rect/theora
/camera/depth_registered/sw_registered/image_rect/theora/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect/theora/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw
/camera/depth_registered/sw_registered/image_rect_raw/compressed
/camera/depth_registered/sw_registered/image_rect_raw/compressed/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/compressed/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/compressedDepth/parameter_updates
/camera/depth_registered/sw_registered/image_rect_raw/theora
/camera/depth_registered/sw_registered/image_rect_raw/theora/parameter_descriptions
/camera/depth_registered/sw_registered/image_rect_raw/theora/parameter_updates
/camera/driver/parameter_descriptions
/camera/driver/parameter_updates
/camera/ir/camera_info
/camera/ir/image
/camera/ir/image/compressed
/camera/ir/image/compressed/parameter_descriptions
/camera/ir/image/compressed/parameter_updates
/camera/ir/image/compressedDepth
/camera/ir/image/compressedDepth/parameter_descriptions
/camera/ir/image/compressedDepth/parameter_updates
/camera/ir/image/theora
/camera/ir/image/theora/parameter_descriptions
/camera/ir/image/theora/parameter_updates
/camera/projector/camera_info
/camera/rgb/camera_info
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
/camera/rgb/image_rect_color
/camera/rgb/image_rect_color/compressed
/camera/rgb/image_rect_color/compressed/parameter_descriptions
/camera/rgb/image_rect_color/compressed/parameter_updates
/camera/rgb/image_rect_color/compressedDepth
/camera/rgb/image_rect_color/compressedDepth/parameter_descriptions
/camera/rgb/image_rect_color/compressedDepth/parameter_updates
/camera/rgb/image_rect_color/theora
/camera/rgb/image_rect_color/theora/parameter_descriptions
/camera/rgb/image_rect_color/theora/parameter_updates
/camera/rgb_rectify_color/parameter_descriptions
/camera/rgb_rectify_color/parameter_updates
/cells_vis_array
/clicked_point
/clusters_centers
/collision_object
/discovered_space
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/feedback_states
/free_cells_vis_array
/initialpose
/joint_path_command
/joint_states
/joint_trajectory_action/cancel
/joint_trajectory_action/feedback
/joint_trajectory_action/goal
/joint_trajectory_action/result
/joint_trajectory_action/status
/menu/feedback
/menu/update
/menu/update_full
/move_base_simple/goal
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/filtered_cloud
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/occupied_cells_vis_array
/octomap_binary
/octomap_full
/octomap_point_cloud_centers
/octomap_server_node/parameter_descriptions
/octomap_server_node/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/point_cloud_filter/points
/point_cloud_filter/rviz/filter
/pose_arrows
/pose_text
/pose_text_array
/projected_map
/ray_cast_lines
/ray_cast_lines_array
/recognized_object_array
/robot_status
/rosout
/rosout_agg
/rviz/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz/motionplanning_planning_scene_monitor/parameter_updates
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/tf
/tf_static
/trajectory_execution_event
/unknown_full_map
/unknown_pc
```

joao-pm-santos96, 2019-06-13:
the ones I think are needed:

```bash
/camera/depth/camera_info
/camera/depth_registered/points
/camera/rgb/camera_info
/camera/rgb/image_raw
/cells_vis_array
/clusters_centers
/discovered_space
/free_cells_vis_array
/joint_states
/occupied_cells_vis_array
/octomap_binary
/octomap_full
/octomap_point_cloud_centers
/planning_scene_world
/point_cloud_filter/points
/point_cloud_filter/rviz/filter
/pose_arrows
/pose_text
/pose_text_array
/ray_cast_lines
/ray_cast_lines_array
/tf
/tf_static
/unknown_full_map
/unknown_pc
```

joao-pm-santos96, 2019-06-13:
> the ones I think are needed:
> 
> ```shell
> /camera/depth/camera_info
> /camera/depth_registered/points
> /camera/rgb/camera_info
> /camera/rgb/image_raw
> /cells_vis_array
> /clusters_centers
> /discovered_space
> /free_cells_vis_array
> /joint_states
> /occupied_cells_vis_array
> /octomap_binary
> /octomap_full
> /octomap_point_cloud_centers
> /point_cloud_filter/points
> /point_cloud_filter/rviz/filter
> /pose_arrows
> /pose_text
> /pose_text_array
> /ray_cast_lines
> /ray_cast_lines_array
> /tf
> /tf_static
> /unknown_full_map
> /unknown_pc
> ```

Maybe not all because my laptop can’t handle it

joao-pm-santos96, 2019-06-14:
I am using this launchfile https://github.com/joao-pm-santos96/SmObEx/blob/master/smobex_bringup/launch/play_bag.launch but the bag allways stays .active at end

joao-pm-santos96, 2019-06-14:
@miguelriemoliveira any idea why it happens?

miguelriemoliveira, 2019-06-14:
Hi João,

I have to look into it. Still here on the test, so probably today it will
be dificult.

Are you still in LAR?

Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

joao-pm-santos96, 2019-06-14:
Yes I’m still here

email: mriem@ua.pt<mailto:mriem@ua.pt> | m.riem.oliveira@gmail.com<mailto:m.riem.oliveira@gmail.com>

miguelriemoliveira, 2019-06-14:
Ok. I will try to pass there when this is over.

joao-pm-santos96, 2019-06-14:
> email: mriem@ua.pt<mailto:mriem@ua.pt><mailto:mriem@ua.pt> | m.riem.oliveira@gmail.com<mailto:m.riem.oliveira@gmail.com><mailto:

Ok, thank you

joao-pm-santos96, 2019-06-14:
done on commit https://github.com/joao-pm-santos96/SmObEx/commit/fb8d227a5fe006bbb49b016078b047ba523d27a6

closed 2019-06-14 by joao-pm-santos96